abandonware/noble #203: Error occurred in handler for 'create-connection': Error: Cannot find module './native/binding'

akiannillo, 2021-04-27:
I'm trying to use noble with electron + webpack but I got this error.

```
Error occurred in handler for 'create-connection': Error: Cannot find module './native/binding'
    at webpackMissingModule (/Users/xxx/Workspace/xxx-app/.webpack/main/index.js:3633:82)
    at Object../node_modules/@abandonware/noble/lib/mac/bindings.js (/Users/xxx/Workspace/xxx-app/.webpack/main/index.js:3633:173)
    at __webpack_require__ (/Users/xxx/Workspace/xxx-app/.webpack/main/index.js:21:30)
    at module.exports../node_modules/@abandonware/noble/lib/resolve-bindings.js.module.exports (/Users/xxx/Workspace/xxx-app/.webpack/main/index.js:4484:17)
    at Object../node_modules/@abandonware/noble/index.js (/Users/xxx/Workspace/xxx-app/.webpack/main/index.js:98:128)
    at __webpack_require__ (/Users/xxx/Workspace/xxx-app/.webpack/main/index.js:21:30)
    at createConnection (/Users/xxx/Workspace/xxx-app/.webpack/main/index.js:19500:17)
    at /Users/xxx/Workspace/xxx-app/.webpack/main/index.js:19528:5
    at electron/js2c/browser_init.js:201:579
    at Object.<anonymous> (electron/js2c/browser_init.js:161:10005) {
  code: 'MODULE_NOT_FOUND'
}

```

blamay, 2021-05-04:
Make sure to add`@abandonware/noble` to [externals](https://webpack.js.org/configuration/externals/) in webpack config

jkeating1, 2021-08-14:
do you have an example showing how this is done? I'm having trouble figuring out what applies to this situation when I google through the webpack documentation.