abandonware/noble #206: "Binding options" commit broke web bluetooth

antlai, 2021-05-02:
I think the commit  "Binding options"  743b9755170 Feb 28 changed "new NobleBindings"  to just return the constructor (lib/webbluetooth/bindings.js), but in the case of webbluetooth nobody calls "new" anymore, and when trying to add a handler to the constructor, it crashes because it is not an event emitter...

The fix is just to add  back the 'new' in lib/webbluetooth/bindings.js, since it does not have binding options...

rzr, 2021-05-02:
What about reverting this change ?

cc: @atrovato

atrovato, 2021-05-02:
No need to revert, fixed with https://github.com/abandonware/noble/pull/207

I am currectly adding a lot of tests, with coverage, to check this king of regressions.

PR #207 (2021-05-02): Fix ctors #206
Fix #206
closed 2021-05-02 by rzr