abandonware/noble #215: Crash on Electron (OSX)

ks75vl, 2021-07-31:
I got a crash report on OSX when exit Electron app.

Electron version: `13.1.7`
OS version: `macOS 11.3.1 (20E241)`

Crash report:
```
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

0   com.github.Electron.framework 	0x000000010c7e82a0 napi_delete_reference + 48
1   binding.node                  	0x0000000111f4a2fc Napi::Reference<Napi::Function>::~Reference() + 25 (napi-inl.h:2503) [inlined]
2   binding.node                  	0x0000000111f4a2fc Napi::FunctionReference::~FunctionReference() + 25 (napi.h:1283) [inlined]
3   binding.node                  	0x0000000111f4a2fc Napi::FunctionReference::~FunctionReference() + 25 (napi.h:1283) [inlined]
4   binding.node                  	0x0000000111f4a2fc ThreadSafeCallback::Impl::~Impl() + 188 (napi-thread-safe-callback-impl.hpp:8)
5   binding.node                  	0x0000000111f4a226 ThreadSafeCallback::Impl::~Impl() + 8 (napi-thread-safe-callback-impl.hpp:8) [inlined]
6   binding.node                  	0x0000000111f4a226 ThreadSafeCallback::Impl::async_callback()::'lambda'(uv_handle_s*)::operator()(uv_handle_s*) const + 18 (napi-thread-safe-callback-impl.hpp:97) [inlined]
7   binding.node                  	0x0000000111f4a226 ThreadSafeCallback::Impl::async_callback()::'lambda'(uv_handle_s*)::__invoke(uv_handle_s*) + 22 (napi-thread-safe-callback-impl.hpp:96)
8   com.github.Electron.framework 	0x00000001093a6ad7 uv_run + 535
9   com.github.Electron.framework 	0x000000010c7d47b1 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 225105
10  com.github.Electron.framework 	0x000000010c7d4c43 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 226275
11  com.github.Electron.framework 	0x000000010c79a773 node::FreeEnvironment(node::Environment*) + 99
12  com.github.Electron.framework 	0x000000010948afc5 ElectronInitializeICUandStartNode + 863333
13  com.github.Electron.framework 	0x0000000109481e29 ElectronInitializeICUandStartNode + 826057
14  com.github.Electron.framework 	0x000000010a688ca2 v8::internal::SetupIsolateDelegate::SetupHeap(v8::internal::Heap*) + 2928546
15  com.github.Electron.framework 	0x000000010a68a2ca v8::internal::SetupIsolateDelegate::SetupHeap(v8::internal::Heap*) + 2934218
16  com.github.Electron.framework 	0x000000010a686120 v8::internal::SetupIsolateDelegate::SetupHeap(v8::internal::Heap*) + 2917408
17  com.github.Electron.framework 	0x0000000109adf2c7 electron::fuses::IsCookieEncryptionEnabled() + 5721719
18  com.github.Electron.framework 	0x0000000109adedc8 electron::fuses::IsCookieEncryptionEnabled() + 5720440
19  com.github.Electron.framework 	0x0000000109addf16 electron::fuses::IsCookieEncryptionEnabled() + 5716678
20  com.github.Electron.framework 	0x0000000109ade002 electron::fuses::IsCookieEncryptionEnabled() + 5716914
21  com.github.Electron.framework 	0x00000001093b8326 ElectronMain + 134
22  com.github.Electron           	0x00000001093467e6 0x109345000 + 6118
23  libdyld.dylib                 	0x00007fff20736f3d start + 1
```

My Electron main process:
```javascript
const { app, BrowserWindow } = require('electron');
const noble = require('@abandonware/noble');

app.whenReady().then(() => {

    let win = new BrowserWindow();

    win.loadURL('https://example.com').then(() => {
        // win.webContents.openDevTools();
        win.maximize();
    });

    noble.on('stateChange', async (state) => {
        if (state === 'poweredOn') {
            await noble.startScanningAsync(['180f'], false);
        }
    });

    noble.on('discover', async (peripheral) => {
        console.log(peripheral.uuid);
    });

    app.on('window-all-closed', async function () {
        await noble.stopScanningAsync();
        app.quit();
    });
});
```

enami, 2021-08-04:
We have same issue here.
It looks like the `env` passed to napi_delete_referece() is already invalidated when the thread deletes `ThreadSafeCallback::Impl` object.

ks75vl, 2021-08-04:
> We have same issue here.
> It looks like the `env` passed to napi_delete_referece() is already invalidated when the thread deletes `ThreadSafeCallback::Impl` object.

Hi enami,
Did you find a way to fix it temporarily?

enami, 2021-08-04:
If you want simple workaround, then just not to delete ThreadSafeCallback::Impl object.  Since an application is going to exit anyway, not deleting this small object seems not so much harmful.  Of course, this is not a correct fix though.

```
--- node_modules/napi-thread-safe-callback/napi-thread-safe-callback-impl.hpp~	2021-08-05 04:53:40.000000000 +0900
+++ node_modules/napi-thread-safe-callback/napi-thread-safe-callback-impl.hpp	2021-08-05 04:54:13.000000000 +0900
@@ -94,7 +94,7 @@
 
             if (close_)
                 uv_close(reinterpret_cast<uv_handle_t *>(&handle_), [](uv_handle_t *handle) {
-                    delete static_cast<Impl *>(handle->data);
+                    // delete static_cast<Impl *>(handle->data);
                 });
         }
 
```

enami, 2021-08-04:
I wonder if a correct fix would be using napi's thread-safe-function instead of using napi-thread-safe-callback npm module.

enami, 2021-08-10:
@ks75vl Hi, please try #219 if possible.  It fixes the crash at least for me.