abandonware/noble #216: macOS Monterey & Big Sur

donavanbecker, 2021-08-04:
we are using this in https://github.com/OpenWonderLabs/node-switchbot and on macOS, just can't seem to get it to work. 

is anyone else having any issues with macOS?

ansoni-san, 2021-08-05:
I can confirm that no matter what signed entitlements and Info.plist entries I add it just doesn't seem to work on Monterey.

Anyone else tried?

ansoni-san, 2021-08-10:
A scan simply never returns any devices. The debug output simple says scan started, and there are no auth / code sign errors shown in `dmesg`.

acstacey, 2021-08-17:
I am also having issues in the Monterey beta, debugging with Xcode only shows that it gets as far as calling the scan function but then no discovery callbacks are called.  The same code is working on Big Sur.

acstacey, 2021-08-17:
I have been able to get scan responses using node 12 and filtering by the advertised service.  But then connection is unreliable.

ebaauw, 2021-10-01:
Using noble on NodeJS v14 (latest LTS version).  It does do something on macOS, but not reliably.  I can only read or write a single characteristic per connected session.  I need to disconnect/reconnect to read/write the next characteristic.  I cannot get characteristic notifications to work at all.  Same code runs without issues on Pi OS.  As far as I can tell, it's an issue with the mac bluetooth binding; peripherals are discovered just fine.

Tried Big Sur (11.6) as well as High Sierra (10.13) - I no longer have any older versions.