abandonware/noble #225: In the latest mac system (maOS Monterey, 12.0.1), there is no response to the discover method

lwj06, 2021-10-20:
In the latest mac system (maOS Monterey, 12.0.1), there is no response to the discover method
[image]
If use an lower version of the macOS, I can discover services and connect bluetooth equipment

bubez81, 2021-11-20:
any news about this problem?

pavelkrcil, 2021-12-03:
Hello, had anyone luck to solve this issue? We are in the same trouble with our application.
Thanks.

onderweg, 2021-12-03:
Same here, was also not able to find any workaround.

davidnixon, 2021-12-15:
This seems to work but still testing ...
```javascript
noble.startScanning(["be15beef6186407e83810bd89c4d8df4"], false);
```
So if you know the service id of what you are looking for it it seem to work. Using this form:
```javascript
noble.startScanning(serviceUUIDs, allowDuplicates[, callback(error)]); // particular UUIDs
```

davidnixon, 2021-12-15:
I feel like it has something to do with this comment about scanning in the background
https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipherals
"your app must explicitly scan for one or more services by specifying them in the serviceUUIDs parameter"

ebaauw, 2021-12-15:
> This seems to work

Doesn't seem to do anything for me, under macOS 12.1.

bwp91, 2021-12-15:
> This seems to work but still testing ...
> 
> ```js
> noble.startScanning(["be15beef6186407e83810bd89c4d8df4"], false);
> ```
> 
> So if you know the service id of what you are looking for it it seem to work. Using this form:
> 
> ```js
> noble.startScanning(serviceUUIDs, allowDuplicates[, callback(error)]); // particular UUIDs
> ```

I can't get this to work either. On macOS 12.1, node 16.13.1, Xcode installed and `node` added to the system bluetooth permissions.

davidnixon, 2021-12-15:
This bit "be15beef6186407e83810bd89c4d8df4" is the service uuid for an Anki Overdrive car. So you need to know the service id of whatever you are trying to discover. Currently looking for a way to find service uuid of other devices ...

ebaauw, 2021-12-15:
> This bit "be15beef6186407e83810bd89c4d8df4" is the service uuid for an Anki Overdrive car. So you need to know the service id of whatever you are trying to discover. Currently looking for a way to find service uuid of other devices ...

And I suppose this service actually needs to be advertised by the device.  The devices I use Noble for don't advertise any services, so this workaround wouldn't help me, even if I got it to work.  Tried several short and one long UUID advertised by my other devices, but no luck.

> Currently looking for a way to find service uuid of other devices ...

I'm using nRF Connect by Nordic Semiconductor on my iPad to discover and probe BLE devices.  You could also run Noble on another platform (Raspberry Pi) and write a small programme to list the received advertisements.

sidecutter, 2021-12-17:
> This seems to work but still testing ...
>
> ```js
> noble.startScanning(["be15beef6186407e83810bd89c4d8df4"], false);
> ```
>
> So if you know the service id of what you are looking for it it seem to work. [...]

Works for me on MacOS 12.0.1. I'm able to discover and connect to the device with the specified (and advertised) service uuid. There seems to be another problem though, as I am now unable to write values to BLE Characteristics. 

MacOS 12.0.1
Node v14.17.6
@abandonware/noble 1.9.2-15

dlech, 2021-12-29:
I hope everyone will report this issue to Apple using the Feedback Assistant app as well.

alin256, 2022-01-09:
Interestingly, discovery works via Chrome. But there are still some problems with the connection.

I am using the library via node-poweredup.
https://github.com/nathankellenicki/node-poweredup/issues/148

suda, 2022-02-03:
I believe [this SO post](https://stackoverflow.com/questions/69661809/no-name-or-address-cbcentralmanager-no-longer-working-on-macos-12) sheds some light on this bug. Apparently it's fixed in macOS 12.3 yet to be released:

> macOS 12.3 Release Date
With regards to iOS 15.4 we have predicted that it won't arrive until April 2022. This is based on the amount of time Apple spent developing the iOS 14.5 beta in 2021.
> Of course we could see macOS 12.3 sooner than iOS 14.5, but it seems likely that they will all arrive in tandem. So an April launch looks likely, but there is a chance we could see the new features in March.

[via MacWorld](https://www.macworld.co.uk/news/macos-123-release-date-features-mac-3812882/)

bw1213, 2022-03-22:
I've confirmed discovery works on MacOS 12.3.

AllanOricil, 2022-04-10:
> I've confirmed discovery works on MacOS 12.3.

@bw1213 can you explain what you did exactly? Im using MacOS 12.3.1 + node 16 and it does not discover any devices. It simpy never calls the discover callback

davidnixon, 2022-04-11:
It works for me too on 12.3.1

This is the workaround I add earlier which limits the scan to just the advertised service uuids in the array. This still works for me and I do not need to change it.
**earlier workaround**
```javascript
 noble.startScanning(
              ["be15beef6186407e83810bd89c4d8df4"],
              false,
              (error) => {
                if (error) debug(error);
              }
            );
```

If you do not know the service uuids then of course you cannot specify them. This code now works for me in 12.3.1. This code did not work back in December but it does look like something in 12.3.1 has fixed it.
**now works again without specifying service uuids**
```javascript
 noble.startScanning(
              [],
              false,
              (error) => {
                if (error) debug(error);
              }
            );
```

davidnixon, 2022-04-26:
Probably this should be closed now.

donavanbecker, 2022-05-15:
Has anyone tried the latest beta, not working for me

onderweg, 2022-05-19:
I can also confirm discovery works on Monterey 12.3.

donavanbecker, 2022-05-19:
What about 12.4 or 12.5? anyone having issues?

Apollon77, 2022-05-29:
no issues anymore for me with 12.4