abandonware/noble #233: peripheral can not connect

wangjianxiang, 2021-11-29:
nodejs ：10.16.0
abandonware/noble：  Release 1.9.2-15

run it question: 
`Debugger attached.
peripheral with ID a09e1a7e4965 found
  Local Name        = Polar
  Manufacturer Data = 6b0072080000
  Service Data      = []
  Service UUIDs     = 156d,feer
services and characteristics:
FATAL ERROR:  The value of "offset" is out of range. It must be >= 0 and <= 3. Received 5
 1: 00007FF6A0BDC6AA v8::internal::GCIdleTimeHandler::GCIdleTimeHandler+4506
 2: 00007FF6A0BB7416 node::MakeCallback+4534
 3: 00007FF6A0BB7D90 node_module_register+2032
 4: 00007FF6A0BB7DA9 node_module_register+2057
 5: 00007FF6A0BAAACF napi_fatal_error+175
 6: 00007FF84CB21887
 7: 00007FF84CB691EE
 8: 00007FF84CB4D570
 9: 00007FF84CB4C4C9
10: 00007FF894A11456 RtlCaptureContext2+1190
11: 00007FF84CB32934
12: 00007FF84CB224F8
13: 00007FF6A0BAA1B7 node::Buffer::New+43767
14: 00007FF6A0C20C78 uv_run+280
15: 00007FF6A0BBE76E node::NewContext+1390
16: 00007FF6A0BBED7B node::NewIsolate+603
17: 00007FF6A0BBF1D7 node::Start+823
18: 00007FF6A0A6F3CC node::MultiIsolatePlatform::MultiIsolatePlatform+604
19: 00007FF6A16B6E3C v8::internal::compiler::OperationTyper::ToBoolean+129516
20: 00007FF892AE7034 BaseThreadInitThunk+20
21: 00007FF8949C2651 RtlUserThreadStart+33

Process finished with exit code 134
`

iJamesPHP, 2022-02-22:
Seeing the same behaviour.

rzr, 2022-02-22:
Can you compare with earlier version ?