abandonware/noble #239: Ubuntu 20.04 and recent-ish intel wireless chips

shigmas, 2022-01-18:
Little test matrix with noble on Ubuntu with some Intel chips
| Ubuntu  |  AC9462 | AX201 |
| 18.04 | Yes | No |
| 20.04 | No | No |
But, with a little hack, I've got it working for both chips on 20.04 by adding a reset() in hci.js, in pollIsDevUp, before setSocketFilter().
 But, this is purely in my tests, by comparing a working application in go and seeing what HCI commands were being sent.

So, I'm of course willing to submit a PR, but I don't know if this is the right place to put it. The only time reset() is called (in hci.js, anyway) is for if _userChannel is set, which still doesn't seem like it's in the right order.

nwaelti, 2022-03-31:
Wow, thanks a lot for that tip, I'm on debian 11 and using a Intel Corporation Wireless-AC 9260 (Asus PCE-AC58BT) and I can confirm, by adding a `this.reset()` before `this.setSocketFilter()` made the whole thing work...

Patronics, 2022-05-03:
@nwaelti, @shigmas, did you have to do anything after editing the file (node_modules/@abandonware/noble/lib/hci-socket/hci.js, I assume)? Like rebuilding noble or anything like that? I'm trying to use a AX200 bluetooth device with ubuntu 22.04, with no success so far, despite adding that reset line as you described. Thanks in advance!

nwaelti, 2022-05-03:
> did you have to do anything after editing the file

Actually not, as soon as the `reset()` was in place, everything started to work. Aside maybe a global restart of the bluetooth subsystem (or a reboot).

shigmas, 2022-05-05:
I don't think anything except starting the application that's using noble is required, but I'm reinstalling the OS at the time, so I wouldn't really know.

I wouldn't be surprised if something else was different in the iwlwifi drivers between 20.04 and 22.04. Best bet is to try another HCI implementation and comparing the calls. Or ditch this abandonware.

geneccx, 2022-07-14:
This worked for me on my 10th gen Intel NUC. Thanks for sharing!

franckOL, 2022-07-21:
I try

> Wow, thanks a lot for that tip, I'm on debian 11 and using a Intel Corporation Wireless-AC 9260 (Asus PCE-AC58BT) and I can confirm, by adding a `this.reset()` before `this.setSocketFilter()` made the whole thing work...

I try that, now able to scan devices but not able to connect to it.
I check with bluetoothctl connection is OK

ianchanning, 2024-04-10:
On an ARM debian 11 compulab gateway (inside a docker container), and this was indeed the fix.

We have forked [noble with this fix in place](https://github.com/charphq/noble/commit/1a2ff44fd3e2d574564bb4cff859224e7fe11b80). 

Then for now we just build using the a direct github package link in `package.json` rather than try to get the whole noble package build process working:

```json
  "dependencies": {
    "@abandonware/noble": "github:charphq/noble#master"
  }
```