abandonware/noble #240: `manufactures-hex.json` missing

smeijer, 2022-01-27:
When using this lib, I get the following error thrown:

```
ERROR:	Could not load manufacturer list from "./manufactures-hex.json".
	Please see "/code/desk/node_modules/@abandonware/noble/lib/../scripts/manufactures/README.md" for instructions.
```

Triggered by this code:

https://github.com/abandonware/noble/blob/2af44d809337e14bb527bca08e5d75ca829365b6/lib/manufacture.js#L2-L8

Except, there is no such README. What am I missing?

rzr, 2022-01-27:
just revert my last change , or maybe just replace error with warnings and review open PR:

https://github.com/abandonware/noble/pulls

Apollon77, 2023-01-05:
Can be closed, code was removed in -19

rzr, 2023-01-19:
Is anyone in touch with BT SIG ? Because I'd like to have them to publish this with compatible license