abandonware/noble #246: FATAL ERROR:  The value of "offset" is out of range. It must be >= 0 and <= 3. Received 5

maximus1127, 2022-03-11:
When I try to connect to a peripheral, I get this error probably a solid 50% of the time. Scanning works, discovery works, but once I go to connect to a peripheral, this error pops up. I've tried to trace it into the source files but with no luck. Try/catch also does not work. Is there a quick fix for this? Or some nuance that I'm unaware of that can resolve this?

[image]

maximus1127, 2022-03-11:
just in case anyone comes here in the future and needs an immediate and quick work around, the way I resolved it quite reliably was to set a variable equal to the peripheral on discovery and stopped the scan. Then gave the user a button to click and sent an ipc message to the main process calling the "connect" method on that variable. It seems to work great when not trying to handle it all in one long, connected string of methods. Sure, my user has to go through one extra step of connecting manually as opposed to having the system do it when it boots the program but that's better than having to open the program three times in a row because it crashes.

gdoumen, 2022-06-13:
The issue seems to be caused by the fact that `this._aclBuffers.length` is 0, when `Hci.prototype.writeAclDataPkt` is being called

File: lib/hci-socket/hci.js, lines 654ff
`
  const l2capLength = 4 /* l2cap header */ + data.length;
  const aclLength = Math.min(l2capLength, this._aclBuffers.length);
  const first = Buffer.alloc(aclLength + 5 /* acl header */);

  // acl header
  first.writeUInt8(HCI_ACLDATA_PKT, 0);
  first.writeUInt16LE(handle | (ACL_START_NO_FLUSH << 12), 1);
  first.writeUInt16LE(aclLength, 3);

  // l2cap header
  first.writeUInt16LE(data.length, 5);
  first.writeUInt16LE(cid, 7);
`

If  `this._aclBuffers.length` is 0, the buffer `first` is allocated with 5 bytes and the l2cap header is not fitting.

gdoumen, 2022-06-13:
I could make my app work by changing 

```
 const first = Buffer.alloc(aclLength + 5 /* acl header */);
```
to 
``` 
  const first = Buffer.alloc(l2capLength+ 5 /* acl header */);
```

but this breaks the tests. 

And as I don't know the code well enough, I don't want to raise a PR with failing test cases

dan2k3k4, 2023-03-13:
Getting similar issue on https://github.com/tonymyatt/zwifit

```
Characteristic not supported: 17
Characteristic not supported: 1
Characteristic not supported: 17
internal/buffer.js:77
  throw new ERR_OUT_OF_RANGE(type || 'offset',
  ^

RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 8
    at boundsError (internal/buffer.js:77:9)
    at Buffer.readUInt16LE (internal/buffer.js:234:5)
    at Object.fromBuffer (/home/dan/zwifit/src/ble/ifit/_constants.js:128:40)
    at /home/dan/zwifit/src/ble/ifit/_request.js:801:45
    at Array.forEach (<anonymous>)
    at parseWriteAndReadResponse (/home/dan/zwifit/src/ble/ifit/_request.js:795:35)
    at /home/dan/zwifit/src/ble/ifit/_request.js:174:18
    at Characteristic.listener (/home/dan/zwifit/src/ble/ifit/_request.js:660:4)
    at Characteristic.emit (events.js:315:20)
    at Noble.onRead (/home/dan/zwifit/node_modules/noble/lib/noble.js:440:20) {
  code: 'ERR_OUT_OF_RANGE'
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zwifit@1.1.0 start: `node app.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zwifit@1.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```