abandonware/noble #248: Add API to "stop" noble

Coding-Kiwi, 2022-03-21:
Original issue: https://github.com/noble/noble/issues/299

Because of the polling here https://github.com/abandonware/noble/blob/master/lib/hci-socket/hci.js#L156 it is not possible to properly stop noble / the process except via `process.exit()`

This PR in the original repository would fix that
https://github.com/noble/noble/pull/577

Meanwhile for anyone who is looking for a workaround:

```js
const noble = require("@abandonware/noble");

/**
 * Currently there is no way to properly shutdown / stop noble.
 * There is an issue/PR but its in the abandoned original `noble/noble` project https://github.com/noble/noble/issues/299
 * The function overwrites / modifications in this file mirror the changes suggested in this PR https://github.com/noble/noble/pull/577
 */

const hci = noble._bindings._hci;

if (hci) {
    hci.stop = function () {
        this._socket.stop();
        clearTimeout(this._pollIsDevUpTimeout);
    };

    hci.pollIsDevUp = function () {
        const isDevUp = this._socket.isDevUp();

        if (this._isDevUp !== isDevUp) {
            if (isDevUp) {
                if (this._state === 'poweredOff') {
                    this._socket.removeAllListeners();
                    this._state = null;
                    this.init();
                    return;
                }
                this.setSocketFilter();
                this.setEventMask();
                this.setLeEventMask();
                this.readLocalVersion();
                this.writeLeHostSupported();
                this.readLeHostSupported();
                this.readLeBufferSize();
                this.readBdAddr();
            } else {
                this.emit('stateChange', 'poweredOff');
            }

            this._isDevUp = isDevUp;
        }

        this._pollIsDevUpTimeout = setTimeout(this.pollIsDevUp.bind(this), 1000);
    };

    noble.stop = function () {
        this._bindings._hci.stop();
    };
}

module.exports = noble;
```

Fefedu973, 2024-05-01:
How to use that in an actual project ? I created a noble.js file and now I import this code instead of noble in other files but when I try to doe noble.stop it dosen't work (not a function)

rzr, 2024-05-01:
Feel free to cherry pick the change and create a PR

Relate-to: https://github.com/noble/noble/pull/577#issuecomment-590051070