abandonware/noble #255: Direct abort when running quick start example from Mac M1

StephaneJuban, 2022-05-04:
Hi,

I am trying to transfert my local server from my old Macbook Intel to my new one running on macOS Monterey and M1 chip.

I am using the same version of node (16.0.0) on both machine.

But on my M1, as soon as I start the quick start example, the server crash.
```
node app.js
[1]    44669 abort      node app.js
```

The crash happens as soon as I reach `noble.on('stateChange', async (state) => { ... });`

Any idea why ?

---

- node v16.0.0
- Macbook M1 Pro
- macOS Monterey 12.3.1

si-stratisiot, 2022-05-20:
I'm getting the same error with macOS 12.4 and Intel chip

squidgetx, 2022-05-22:
I was able to solve this on 12.3.1 Intel mac by adding terminal to allowed bluetooth apps using these instructions: https://github.com/abandonware/noble#os-x

StephaneJuban, 2022-05-23:
> I was able to solve this on 12.3.1 Intel mac by adding terminal to allowed bluetooth apps using these instructions: https://github.com/abandonware/noble#os-x

Thanks this worked !

closed 2022-05-23 by StephaneJuban