abandonware/noble #262: Can not discover peripherals on Ubuntu 22.04

lmarschall, 2022-07-10:
Hey folks,

at the moment, I can't figure out, why my noble script does not work properly.
I'm on Ubuntu 22.04 and Node 16 trying to scan for my iPad next to my laptop.
With `hcitool` I was able to find the tablet, but not with the noble script.

The logs:
```
1552 Source/wifi-bluetooth-bridge git:(main) ✗ » hcitool scan 
Scanning ...
        4C:32:75:76:53:BA       iPad von Lukas
1553 Source/wifi-bluetooth-bridge git:(main) ✗ » npm run serve

> wifi-bluetooth-bridge@1.0.0 serve
> tsc && sudo node dist/app.js

Start scanning...
server is listening on 8080
stateChange
poweredOn
```

Here's my code:
```
// const noble = require('@abandonware/noble')({extended: false});
const noble = require('@abandonware/noble');

export function start() {
  // noble.reset()
  console.log("Start scanning...");
  // noble.startScanning();
}

noble.on('stateChange', async (state) => {
    console.log('stateChange');
    // await noble.startScanningAsync();
    if (state === 'poweredOn') {
        console.log('poweredOn');
        // await noble.startScanningAsync(['180f'], false);
        await noble.startScanningAsync([], true);
    }
});

noble.on('scanStop', async () => {
  console.log("Scanning stopped");
});

noble.on('discover', async (peripheral) => {
    console.log('discover');
    await noble.stopScanningAsync();
    await peripheral.connectAsync();
    const {characteristics} = await peripheral.discoverSomeServicesAndCharacteristicsAsync(['180f'], ['2a19']);
    const batteryLevel = (await characteristics[0].readAsync())[0];

    console.log(`${peripheral.address} (${peripheral.advertisement.localName}): ${batteryLevel}%`);

    await peripheral.disconnectAsync();
    // process.exit(0);
});
```

closed 2022-07-14 by lmarschall

fensterhebel, 2023-02-26:
What was the solution?

lmarschall, 2023-02-26:
I used another package, https://github.com/chrvadala/node-ble :)

rzr, 2023-03-03:
Feel free to rebase foreign changes on @abandonware

lmarschall, 2023-03-04:
Comparing the two repos, I figured that there are baked by different technologies, `node-ble` is using bluez with dbus, `noble` something different I guess. Not sure how compatible changes would be here :(