abandonware/noble #282: v1.9.2-18 v1.9.2-17 are BREAKING!

Apollon77, 2022-12-19:
see https://github.com/abandonware/noble/commit/b67eea246f719947fc45b1b52b856e61637a8a8e#r93600733

Apollon77, 2022-12-19:
The version should be unpublished or a -19 is needed which reverts this brteakting behaviour.

After this we need to propare a major release

bwp91, 2022-12-19:
not sure whether these issues are related, but these have arisen from latest version:

https://github.com/bwp91/homebridge-govee/issues/394

https://github.com/bwp91/homebridge-govee/issues/393

Apollon77, 2022-12-19:
The first one is a rapsi crtash on the build of a module ... no idea why ... maybe too less ram? Other topic. 

But the second issue is excatly this

jins-tkomoda, 2022-12-19:
It doesn't seem to be working on Mac OS either,,,

bwp91, 2022-12-20:
i do not know this package well enough to be able to decipher what needs reverting

bwp91, 2022-12-20:
i’m only commenting on behalf of a lot of my plugin users who are having trouble with the new version 🙂

Apollon77, 2022-12-20:
@kele6ra Can you have a look? It was your PR caused it, so maybe you have an idea how to ideally do it without loosing the feature?

A fallback way in the using libraries would be kind of

[image]

That should work for both cases

Apollon77, 2022-12-20:
@jins-tkomoda Can yiou be more specific WHAT is not working? Also the "noble.on is not a function"? Then yes, this will not work on all OS

kele6ra, 2022-12-20:
@Apollon77 

I have worked with Node 14 lts version then. And have checked it with 16 lts. Now there is 18 version of NodeJS, maybe they have changed something with require call?
I will check it.

kele6ra, 2022-12-20:
We can use another variant:

```
const Noble = require("@abandonware/noble/lib/noble");
const HCIBindings = require("@abandonware/noble/lib/hci-socket/bindings");

const params = {
  deviceId: 0,
  userChannel: true,
  extended: true,
};

const noble = new Noble(new HCIBindings(params));
```

Apollon77, 2022-12-20:
@kele6ra Noooo it has nothing to do with the nodejs version ... Your code change is the reason.

Before: 
```
module.exports = new Noble(bindings);
```

After:
```
module.exports = function (options) {
  return new Noble(bindings);
}
```

Before the instance was returned directly ... now a function is returned that returns the instance ... Thats the issue

kele6ra, 2022-12-20:
I have checked this noble/index.js:

`module.exports = function (options) {
  const Noble = require('./lib/noble');
  const bindings = require('./lib/resolve-bindings')(options);

  return new Noble(bindings);
};`

and it works without any problem.

Apollon77, 2022-12-20:
@kele6ra Yes it works ... this is not the issue but it is a breaking change in using the library!!

Before it was used like

```
const noble = require('...');
noble.on('...', function ...)
```

This is now not working any more and produces the error "noble.on is not a function".

Now you need to use it like 

```
const noble = require('...')({ extended: false });
noble.on('...', function ...)
```

With this we broke the usage of the library release wise in a "pre-release" style version

kele6ra, 2022-12-20:
Yes, but in that moment I haven't found how to transfer parameter to the hci-socket bindings. Because we call for it with the require noble. 
We can add env extended parameter as well. I can update it if you want.

Apollon77, 2022-12-20:
I see the issue. could we go with standard and add a "setBinding" method to the nobiel instace to set the "extended binding" directly afterwards? I did not checked code so far to see if this is an issue

Apollon77, 2022-12-20:
(ind the end we could also revert the whole commit, release a -20 and then add again and release a 2.0

kele6ra, 2022-12-20:
I think that we need to make call modification, because I don't like my solution too =) I have made it because I didn't have enough time then.

kele6ra, 2022-12-20:
Give me 2-3 hours, I need to think how to improve it in better way.

kele6ra, 2022-12-20:
We can just remain extended functions activating with this variant:
```
const HCIBindings = require('@abandonware/noble/lib/hci-socket/bindings');
const Noble = require('@abandonware/noble/lib/noble');

const params = {
  deviceId: 0,
  userChannel: true,
  extended: false
};

const noble = new Noble(new HCIBindings(params));
```

jins-tkomoda, 2022-12-20:
@Apollon77 

> @jins-tkomoda Can you be more specific WHAT is not working? Also the "noble.on is not a function"? Then yes, this will not work on all OS

Yes. "noble.on is not a function" occurs.

kele6ra, 2022-12-20:
@Apollon77 is it okay to set the ble5 extended functional with additional calls, like with choosing another devId?

Apollon77, 2022-12-20:
Ok I could think about two options out of my head

1.) return Nobile instance and add a "setBinding(newBinding)" (or changeBinding(options) which calls resolve.bindings with options anew) method that checks that this.initialized is false and then just overrides this._bindings. if this.initialized is true I would throw an error because then someone tries to set bindungs too late

2.) revert index.js to old version and add a new "with-custom-binding.js" with the new one. So anyone that does ```
require('@abandonware/noble');
```

 gets the original one and for the new option you do 

```
require('@abandonware/noble/with-custom-binding')({ ... });
```

kele6ra, 2022-12-20:
> gets the original one and for the new option you do
> 
> ```
> require('@abandonware/noble/with-custom-binding')({ ... });
> ```

I think, that this variant will be better for future customizing.

Debenben, 2022-12-20:
or

3) use environment variable like

NOBLE_REPORT_ALL_HCI_EVENTS
NOBLE_HCI_DEVICE_ID
NOBLE_MULTI_ROLE
NOBLE_WEBSOCKET
NOBLE_DISTRIBUTED

Apollon77, 2022-12-20:
I would prefer a code solution because Environment variables only make sense if you want to allow code to be controlled from the pitside and honestly this makes only sense in very rare cases when it comes to a "multi purpose library". If an application wants to allow "the user to customize it that way using environment" then this is fine, but I do not see this as a good idea for a library basis

kele6ra, 2022-12-20:
@Apollon77 Should I make pr or you will make it?

Apollon77, 2022-12-20:
If you could do would be awesome. Only on mobile right now. But I can review

kele6ra, 2022-12-20:
https://github.com/abandonware/noble/pull/283

closed 2023-01-05 by Apollon77