abandonware/noble #285: Characteristic write not working on MacOS Ventura

theficus, 2022-12-31:
Environment: MacOS 13.1, @abandonware/noble 1.9.2-15, Node.js 18.12.1

I've got a relatively simple script that largely follows the sample that I'm using to connect to a SensorPush sensor to read temperatures. I can scan, connect, and read values from the device without problem. The problem is that a write operation is required to refresh the temperature.

I have some simple verification code:
```js
let oldValue = await characteristic.readAsync();
await characteristic.writeAsync(someBuffer, true);
await delay(1000); // sleep 1 second
let newValue = await characteristic.readAsync();
```

Whenever I'd run this code, oldValue and newValue would always be identical which isn't expected. I tested this read/write/read using some standalone BLE applications like BlueSee and LightBlue to confirm this works as expected.

I've tried various workarounds such as using discoverSomeServicesAndCharacteristics(Async), discoverServices(Async)/discoverCharacteristics(Async). I've tried sync/async variants. I've also verified when using callbacks or listeners that the write event is emitted, but it never seems to be sent to the device. After messing around this with hours, I tried running the exact same code on a Raspberry Pi (Debian 11 / Bullseye, @abandonware/noble 1.9.2-15, Node.js 18.12.1) and it worked as expected! 

The only logical conclusion I can reach at this point is that there's a problem specific to Mac OS. I'm running Ventura (13.1).

Please let me know if there's more information I can collect. While I ultimately plan to run this node.js service on a Raspberry Pi, it's much nicer doing the actual development work on my Mac. :)

HiveTechDev, 2023-05-31:
I will also bump this. Mac OS and iOS are explicit with the enforcement of write vs write w/o response. you have to call the correct one matching to you peripheral

RootDev4, 2024-08-20:
For me, set writing without response to false did the trick (macOS Sonoma 14.6.1)
`await characteristic.writeAsync(Buffer.from('abc', 'ascii'), false)`