abandonware/noble #293: DBUS suppor

Travelbacon, 2023-01-08:
I am very pleased to see that noble is maintained here.

I was wandering are there any plans to migrate HCI to D-BUS for Linux? That is for now the default way to interact with Bluetooth.

Also, do you guys cash the donations that is being send to the dead main project Noble?
https://opencollective.com/noble People are still donating, while all of their repos are dead.

rzr, 2023-01-09:
thx, maybe it's time to remove opencollective links in readme file because it's upstream one which is not reponsive anymore , does any one know upstream ?

jrobeson, 2023-06-16:
I'ts hci still. Somebody would have to integrate this https://github.com/chrvadala/node-ble or a library like it to use dbus.

rzr, 2023-06-17:
Idealy both can be supported , patches  welcome

Travelbacon, 2023-06-17:
Sorry, not from my side any time. I switched to WiFi for my project. BLE have to much troubles and lost its advantages.