abandonware/noble #312: Ubuntu 22.04 Behavior

vicatcu, 2023-08-03:
Hi and _thank you_ for keeping this library active! I'm using Ubuntu 20.04 (and have now updated to Ubuntu 22.04 as I thought it might help, but it didn't). It's a System76 laptop with Bluetooth integrated.

There are a few quirks I'm dealing with. The first and most unavoidable is that, I can't seem to get past the `await noble.startScanningAsync();` call unless I open the Bluetooth Settings in Ubuntu at which point the call magically completes. I think I did all the stuff that README.md indicates I need to do for Linux/Ubuntu, so I think this is pretty strange. Is there an explanation / workaround for that?

Second unavoidable problem I'm facing is once I _do_ get past that call, the `noble.on('discover', async (peripheral) => {` callback never fires unless I manually disable/enable the Bluetooth in Ubuntu Preferences, sometimes a couple of times (maybe it's a timing issue, I don't know). Is there an explanation / workaround for that?

Happy to share more details with guidance on how to collect those logs / configuration details.

ninjadev64, 2023-09-20:
I'm also using Ubuntu 22.04, having some strange issues. Nothing like what you have, but `await peripheral.connectAsync()` never completes... been wrestling with this library for hours!

sevenbitbyte, 2024-03-12:
After a bit of testing it does not seem like this is an OS bug (for me at least).

I was able to get a USB CSR4 dongle to work just fine on the same amd64 Ubuntu 22.04 system. However explicitly setting the Intel 6E ax1675's USB interface continues to be flaky and return no data. So there seems to be something hardware or driver related at play possibly. The other possibility is that this has something to do with BLE 5.x support as the Intel adapter is pretty new.