abandonware/noble #313: Installation failing on Windows 11 Pro (error C2039 in winrt)

AnnoyingMoose, 2023-09-26:
```
C:\Users\<me>>npm -g inst @abandonware/noble
npm WARN cleanup Failed to remove some directories [
npm WARN cleanup   [
npm WARN cleanup     'C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\@abandonware\\noble',
npm WARN cleanup     [Error: EPERM: operation not permitted, rmdir 'C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\node_modules\@abandonware'] {
npm WARN cleanup       errno: -4048,
npm WARN cleanup       code: 'EPERM',
npm WARN cleanup       syscall: 'rmdir',
npm WARN cleanup       path: 'C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\@abandonware\\noble\\node_modules\\@abandonware'
npm WARN cleanup     }
npm WARN cleanup   ]
npm WARN cleanup ]
npm ERR! code 1
npm ERR! path C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c node-gyp-build
npm ERR! Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
npm ERR!   nothing.c
npm ERR!   win_delay_load_hook.cc
npm ERR!   nothing.vcxproj -> C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\Release\\nothing.lib
npm ERR! cl : command line warning D9025: overriding '/std:c++17' with '/std:c++latest' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR!   noble_winrt.cc
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(983,26): error C2039: 'wait_for': is not a member of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(103): message : see declaration of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(985): message : see reference to class template instantiation 'winrt::impl::consume_Windows_Foundation_IAsyncAction<D>' being compiled [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(1004,26): error C2039: 'wait_for': is not a member of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(103): message : see declaration of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(1006): message : see reference to class template instantiation 'winrt::impl::consume_Windows_Foundation_IAsyncActionWithProgress<D,TProgress>' being compiled [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(1038,26): error C2039: 'wait_for': is not a member of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(103): message : see declaration of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(1040): message : see reference to class template instantiation 'winrt::impl::consume_Windows_Foundation_IAsyncOperationWithProgress<D,TResult,TProgress>' being compiled [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(1057,26): error C2039: 'wait_for': is not a member of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(103): message : see declaration of 'winrt::impl' [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\cppwinrt\winrt\impl\Windows.Foundation.0.h(1059): message : see reference to class template instantiation 'winrt::impl::consume_Windows_Foundation_IAsyncOperation<D,TResult>' being compiled [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\lib\win\src\notify_map.h(31,54): error C2064: term does not evaluate to a function taking 1 arguments [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\lib\win\src\notify_map.h(32,61): error C2064: term does not evaluate to a function taking 1 arguments [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\lib\win\src\notify_map.h(33,1): error C3536: 'serviceHash': cannot be used before it is initialized [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\lib\win\src\notify_map.h(34,1): error C3536: 'characteristicHash': cannot be used before it is initialized [C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble\build\lib\win\binding.vcxproj]
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.4.0
npm ERR! gyp info using node@20.6.0 | win32 | x64
npm ERR! gyp info find Python using Python version 3.11.5 found at "C:\Python311\python.exe"
npm ERR! gyp info find VS using VS2019 (16.11.34031.81) found at:
npm ERR! gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools"
npm ERR! gyp info find VS run with --verbose for detailed information
npm ERR! gyp info spawn C:\Python311\python.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'msvs',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\@abandonware\\noble\\build\\config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\<me>\\AppData\\Local\\node-gyp\\Cache\\20.6.0\\include\\node\\common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=C:\\Users\\<me>\\AppData\\Local\\node-gyp\\Cache\\20.6.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\<me>\\\\AppData\\\\Local\\\\node-gyp\\\\Cache\\\\20.6.0\\\\<(target_arch)\\\\node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\@abandonware\\noble',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'C:\\Users\\<me>\\AppData\\Roaming\\npm\\node_modules\\@abandonware\\noble\\build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Current\Bin\MSBuild.exe
npm ERR! gyp info spawn args [
[... cut ...]
npm ERR! gyp ERR! cwd C:\Users\<me>\AppData\Roaming\npm\node_modules\@abandonware\noble
npm ERR! gyp ERR! node -v v20.6.0
npm ERR! gyp ERR! node-gyp -v v9.4.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in: C:\Users\<me>\AppData\Local\npm-cache\_logs\2023-09-26T13_59_59_049Z-debug-0.log
```

All prerequisites are installed; however, MS Build Tools is installed using the MSVS installer instead of npm. (`npm install --global --production windows-build-tools` also fails.) I have retained the verbose log file and can provide its contents, if requested.

AnnoyingMoose, 2023-09-26:
Discovered that this issue is (sort of?) a duplicate of #277. As @jins-tkomoda suggested, I tried installing the latest Windows SDK and - sure enough - it works now.

Steps to fix (using Visual Studio Installer):
1. Start *Visual Studio Installer*
2. Modify *Visual Studio Build Tools 2019*
3. Switch to *Individual components* tab
4. Select *Windows 11 SDK* (search for "Windows 11 SDK")
5. Download and install
6. Run `npm -g inst @abandonware/noble`

closed 2023-09-26 by AnnoyingMoose

fullmooooon, 2023-11-29:
I really want everyone who encounters this problem to see your answer quickly.
There is nothing more difficult than letting simple nodejs developers like me understand C++errors.
Thank you for making problem-solving easier