abandonware/noble #317: New maintainer in town (bluetooth-hci-socket, noble, bleno) - new features

stoprocent, 2023-11-02:
Hi Guys

I was trying to contribute to abandonware repos but it was taking some time to do so I've created a forks with a few items that were bugging me.

1. Precompiled versions are working (I've fixed github workflows to test and release including precompiled versions)
2. I have added support for HCI over UART to https://github.com/stoprocent/node-bluetooth-hci-socket/#uartserial-any-os so now you can enjoy full HCI support on any OS including MacOS with an access to proper MAC addresses.  There is support with precompiled firmware for `ESP32` and `NRF52`
    - ESP32 - https://github.com/stoprocent/node-bluetooth-hci-socket/tree/main/misc/esp32
    - NRF52 - https://github.com/stoprocent/node-bluetooth-hci-socket/tree/main/misc/nrf52840
4.  With this I had to make some adjustments and fixes for noble which is now also providing precompiled binaries for each os -  https://github.com/abandonware/bleno
5.  And also bleno which same as noble can be now used on Any OS with the UART/Serial dongles - https://github.com/abandonware/bleno
6. I'm happy to contribute all of this back here but I would be nice to become a maintainer so  I can move things faster :)

Im open for requests

Apollon77, 2023-11-02:
Hi @stoprocent,

Cool ... I would love to see all this here ...   @rzr ... next one in the queue ;-))
I also still await the merge of my Maintainer request https://github.com/abandonware/noble/pull/289

@stoprocent mybe also make sue a PR. This was requested by @rzr from me back then.

Ingo

donavanbecker, 2023-11-02:
@stoprocent, why not just open a Pull Request?

Apollon77, 2023-11-03:
@stoprocent maybe one question for my understanding: You had a PR in the hci socket and closed it with comment "I have to change my approach a bit" ... I assumed this was because some technical stuff you need to change, so I was also waiting for another PR on this.

While I agree that things tend to be not that fast here because of @rzr is the only active owner of this Repo right now and it sometimes need a poke more to get things merged (in fact you got a review already before your cleanups...) it basically works. I would also love to see and get the project more active again - also contacting the original devs to take over the original repository did not really worked out.

In my personal(!) view becoming a maintainer for a repo requires trust and you can only build trust by behaviour ... so normally the PR way is the best way to show interest and commitment in a project - and to build that trust. And I know that it works that way because I did that the last 7+ years that way. Yes it is a bit lengthy but it makes sure that all parties feel good about it.

stoprocent, 2023-11-03:
@Apollon77  that's a fair point and I agree 100%. 

This merge "operation" is super complex tho. I need to merge HCI socket first in order to update Noble and Bleno. Then I took an approach for commit messages from angular and this changes how repos should be organized going forward. I've changed all of the workflows so all of this is going to take weeks at the current pace and we are using all of this in the production testing @ ASSA ABLOY and I really didn't have time to wait. I'm happy to create all the merge requests but I don't have time to go back and forth for weeks. I cannot run two parallel tracks one for merge requests with different versioning and naming and the second one for our needs. If I can get someone to help me push this through I will contribute 100%.

Apollon77, 2023-11-03:
Hi and thank you for your honest answer,

I already noticed that you changed more then just code and I'm also honest that this makes it challenging because from the view of the pure project these changes were not needed (they just bring some convenience here and there like the commit.-messages for changelog automation). And yes this makes it now challenging to move stuff over without also changing this completely for here as "upstream" project where also other developers have their opinions how things should be organized. I completely understand that this might have been "easier" but the more you change unrelated things the more you create a forced fork, but in fact too late now ... but yes can all be cleaned up with effort.

Exactly time for such efforts is the main issue ... I got caught very deep in JavaScript implementation of the new matetr standard, but I also rely on noble, bleno and such in several projects so there is still an interest to have a clean and working solution. But all time I invest here I can not make progress on other projects ... thats my dilemma ...

Lets see what @rzr says ...

stoprocent, 2023-11-05:
I've used the same patterns for commits and workflows as in very active JS projects with native bindings like e.g. `serailport`.

Apollon77, 2023-11-06:
Re commit message pattern ... all fine .. But it is still a good example of "different flavours and opinions" :-)

Sooo, and now it came as it needed tocome ... ggrmmppffff ...

Python 3.12 broke/will break a lot nodejs builds as soon as this version goes out. The node-gyp guys now have a new version out but normally npm bundles specific versions, so all npm versions before 10.2.2 are kind of broken as soon as Pythong 3.12 is installed. This will get a huge mess.
And because of the factthe GitHub already updated theor macos runners we are alreayd in the problem world.

So I started to day with node-bluetooth-hci-socket and will push put new versions of all 3 projects with a "node-gyp in the deps" (yes ou id the same, I saw after I did these changes to other projects I maintain too). I also neded to fix Buffer usage (as you also did - see below for a comment).

I will work with @rzr to get them out including now breaking the versioning scheme of the original projects.

We can see how we can work after that to bring your feature changes back in here.

Re Buffer: I think you did two errors:
https://github.com/stoprocent/node-bluetooth-hci-socket/blob/main/lib/usb.js#L245 and https://github.com/stoprocent/node-bluetooth-hci-socket/blob/main/lib/usb.js#L285 need both be "Buffer.from" instead Buffer.alloc!!

stoprocent, 2023-11-07:
@Apollon77 thanks for spotting this: https://github.com/stoprocent/node-bluetooth-hci-socket/releases/tag/v1.2.1

rzr, 2023-11-12:
Hi sorry for the late feedback. I have described some procedure to make this org scalable, I don't want to be the one in charge of trust that's the reason I proposed to offlne this task to existing oss orgs.

See:

https://abandonware.github.io/

https://github.com/abandonware/abandonware.github.io/issues

Apollon77, 2023-11-13:
@rzr see my statement in https://github.com/abandonware/abandonware.github.io/issues/17 ...