abandonware/noble #318: Link Node JS and Node "File exists"

ukiews, 2023-11-16:
When running:

`sudo ln -s /usr/bin/nodejs /usr/bin/node`

per https://github.com/abandonware/noble#prerequisites

I get:
`ln: failed to create symbolic link '/usr/bin/node': File exists`

Is this normal?
I am still trying to get BLE to work with the SwitchBot plugin on the RaspberryPi4

rzr, 2023-11-16:
This is a bad practice this should be changed in the documentation, which OS are you running ?

ukiews, 2023-11-16:
> This is a bad practice this should be changed in the documentation, which OS are you running ?

Homebridge 1.7

donavanbecker, 2023-11-16:
@ukiews, @rzr is asking for your Operating System not app that is using noble.

ukiews, 2023-11-17:
Sorry, I think you are looking for this: 
Raspbian GNU/Linux Bullseye (11)
I'm running a standalone Homebridge server using the Raspberry Pi Imager.

rzr, 2023-11-17:
Please update documentation accordingly  something like ( /usr/bin/node --version || sudo ln -s /usr/bin/nodejs /usr/local/bin/node)

 if you want to use a different node than the default  one,  may you look at update-alternatives command too