abandonware/noble #325: Wrong node-bluetooth-hci-socket version

vwang4536, 2024-02-09:
In the package.json optional dependencies:  "@abandonware/bluetooth-hci-socket": "^0.5.3-11" is expected however the latest version that is currently out for it is ^0.5.3-10. Can we get a quick update to change the package.json to 0.5.3-10?

closed 2024-02-09 by vwang4536

lpgera, 2024-02-09:
@vwang4536 Why did you close the issue? I'm having the same problem and I can't find a solution, so I'd be curious to know yours.

vwang4536, 2024-02-09:
> @vwang4536 Why did you close the issue? I'm having the same problem and I can't find a solution, so I'd be curious to know yours.

I'm just using 1.9.2-23 for now since the version bump for bluetooth-hci-socket was pretty recent. Can also create a PR to revert the version update for bluetooth-hci-socket back to 0.5.3-10

lpgera, 2024-02-10:
If there's no solution then could you reopen the issue please?