abandonware/noble #329: noble not returning correct data

mteichtahl, 2024-02-21:
Hi all

I have found an interesting issue. When I grab the manufacturerData from a peripheral, noble is giving me what appears to be incorrect data. Looking at the two sample, you will see the data from noble is different from the data returned from bluetoothcli. I believe the data from bluetoothcli to be **correct**.

my environment is ```Linux jura 5.15.0-1046-raspi #49-Ubuntu SMP PREEMPT Thu Jan 18 12:45:41 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux``` on ```Ubuntu 22.04.4 LTS```

any guidance would be greatly appreciated

** from noble **
```js
{
  localName: 'TT214H BlueFrog',
  txPowerLevel: undefined,
  manufacturerData: <Buffer ab 00 2a 05 08 03 31 3d e5 2e fa 00 46 43 83 42 01 c0 01 b6 28 00 00 00 00 00 00 00 00>,
  serviceData: [],
  serviceUuids: [ '1523', '1623' ],
  solicitationServiceUuids: [],
  serviceSolicitationUuids: []
}
```

** from bluetoothcli **
```bash
^Croot@jura:/home/amazon/btEngine# bluetoothctl info F8:FB:C3:2A:78:70
Device F8:FB:C3:2A:78:70 (random)
        Name: TT214H BlueFrog
        Alias: TT214H BlueFrog
        Paired: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (00001523-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (00001623-0000-1000-8000-00805f9b34fb)
        ManufacturerData Key: 0x00ab
        ManufacturerData Value:
  2a 05 08 03 31 3d e5 2e fa 00 46 43 83 42 01 c0  *...1=....FC.B..
  01 b6 28 00 00 00 00 00 00 00 00                 ..(........   
```

the relevant code section is below

```ts
noble.on('discover', async (peripheral: Peripheral)=> {
      if (peripheral.address === CoffeeMachineUtils.toHexString(self.machineId)) {
        self.machine = peripheral;
        
        try {
          self.setState(MACHINE_STATE.CONNECTING)
          await peripheral.connectAsync();
          self.setState(MACHINE_STATE.CONNECTED)
          noble.stopScanning();
          console.log(peripheral.advertisement.manufacturerData)
        } catch (e) {
          self.setState(MACHINE_STATE.DISCONNECTED)
          throw e;
        }
      }
    })
```

rzr, 2024-02-22:
Is this reproductible with other devices ?

mteichtahl, 2024-02-22:
It is. Yes. Both with other Bluetooth devices and central devices. We tested with both a pi and osx - same results Please note: My working hours may not be your working hours. Please do not feel obligated to reply outside of your normal work schedule!Sent on the go - please excuse mistakes

rzr, 2024-02-23:
Does this relate to other problem observed elsewhere:

https://github.com/abandonware/bleno/issues/23

mteichtahl, 2024-02-23:
Thanks for the tip. I’ll work through this and try swapping to raspbian

mteichtahl, 2024-02-26:
We have now tried raspbian as well - it seems we are seeing the same behaviour

rzr, 2024-03-08:
Do you know if it's a regression ? I would compare to earlier versions ?