abandonware/noble #340: Add UWP bindings

finaldzn, 2024-08-08:
Hello,

I've been using Noble for quite some time now, but I've encountered challenges using it with Windows. The primary issue is the need to change the Bluetooth drivers, which disrupts the functionality of other Bluetooth devices like mice, keyboards, and headsets.

A library called [noble-uwp](https://github.com/tylerc/noble-uwp/tree/fix-nodejs-12.x) has been developed to address this issue by utilizing the bindings provided by Windows. This approach eliminates the need to change the default Bluetooth driver on Windows.

I would be glad to assist with a potential merge of the work done on noble-uwp. However, I'm not sure where to begin. Additionally, the library is currently only compatible with Node 12.X, which presents a challenge when integrating more libraries into a program.

Please let me know if this is something you might consider. I am more than willing to help with the process.

For more details on the challenges I've faced, you can check [this issue](https://github.com/jasongin/noble-uwp/issues/88).

Thank you.

rzr, 2024-08-08:
> I would be glad to assist with a potential merge of the work done on noble-uwp. However, I'm not sure where to begin.

I would start by checking/adding windows builders to this project, if possible.

Any windows users around to guide your contribution ?

finaldzn, 2024-09-03:
Hello,

Sorry just coming back on this.

I am a windows user.

So what you are meaning is to first add windows support to this project ? Or at least make sure its working ?

By Windows builder, do you also mean updating the CI to create a windows distribution ?

Thanks