abandonware/noble #341: Not get "poweredOn" event in Ubuntu

woowu, 2024-08-17:
Hi,

Since a week ago, my small noble application cannot work in my Ubunut, which can scan and connection BLE devices before.  But for unknown reason, without any code change in my application, it can no longer work and I found it's because it did not the the "poweredOn" event. But the Bluetooth functionality from the OS point of view is working fine because Ubuntu's bluetooth menu can list and connect to devices as before. 

Ubuntu version: 20.04 LTS
Noble version: 1.9.2-25.

What kind problem could it possible be? Where can I find some debug/log information? Thanks.

Kr,
woody

ollivierv, 2024-08-17:
I had the same issue with "poweredOn" event that never happen. But I'm on Ubuntu 24.04. 
And exactly same code is running well on Ubuntu 22.04 on another machine. 
Of course, the two configuration have perfectly operationnal HCI.

woowu, 2024-08-18:
@ollivierv Thanks for your info.  I always on same machine and it's always Ubuntu 20.04 LTS.  Just stopped working at early this before, before that, it always works. I am afraid it's some dependencies on my system has changed but cannot recall what's that.

Where do you get these HCI level log information? Can you teach me? Thanks.

-woody

rzr, 2024-08-19:
Please try older version to find if it is regression or not ?

MrShinyAndNew, 2024-10-08:
I'm having the same problem in Fedora 40. The sample program in the quickstart example just doesn't work for me at all.

I've verified with `bluetoothctl` and `scan le` that BLE scanning is working and can see my bluetooth device (a Lego PoweredUp remote). 

However in node-poweredup nothing happens, and in the noble same program I never receive the poweredOn event or any events at all.

This program prints 'Starting' and then just waits forever.

```
const noble = require('@abandonware/noble');
console.log('Starting');
noble.on('stateChange', async (state) => {
    console.log('stateChange ', state);
  if (state === 'poweredOn') {
    console.log('Powered on');
    await noble.startScanningAsync(['180f'], false);
  }
});

noble.on('discover', async (peripheral) => {
  await noble.stopScanningAsync();
  await peripheral.connectAsync();
  const {characteristics} = await peripheral.discoverSomeServicesAndCharacteristicsAsync(['180f'], ['2a19']);
  const batteryLevel = (await characteristics[0].readAsync())[0];

  console.log(`${peripheral.address} (${peripheral.advertisement.localName}): ${batteryLevel}%`);

  await peripheral.disconnectAsync();
  console.log('Exiting');
  process.exit(0);
});
```

MrShinyAndNew, 2024-10-09:
I set `export DEBUG="*"` and added a console.log and found there's an error setting the HCI filter

  hci setting filter to: 1600000020c10800000000400000 +0ms
  hci onSocketError: EINVAL, Invalid argument +1ms
Error: EINVAL, Invalid argument
    at Hci.setSocketFilter (/home/mark/code/powered_up/powered_up/node_modules/@abandonware/noble/lib/hci-socket/hci.js:262:16)
    at Hci.pollIsDevUp (/home/mark/code/powered_up/powered_up/node_modules/@abandonware/noble/lib/hci-socket/hci.js:184:12)
    at Hci.init (/home/mark/code/powered_up/powered_up/node_modules/@abandonware/noble/lib/hci-socket/hci.js:165:10)
    at NobleBindings.init (/home/mark/code/powered_up/powered_up/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:100:13)
    at /home/mark/code/powered_up/powered_up/node_modules/@abandonware/noble/lib/noble.js:62:26
    at process.processTicksAndRejections (node:internal/process/task_queues:77:11) {
  errno: 22,
  code: 'EINVAL',
  syscall: 'setsockopt'
}
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +6ms
  hci set le event mask - writing: 010120081f00000000000000 +1ms
  hci read local version - writing: 01011000 +0ms
  hci write LE host supported - writing: 016d0c020100 +1ms
  hci read LE host supported - writing: 016c0c00 +0ms
  hci le read buffer size - writing: 01022000 +0ms
  hci read bd addr - writing: 01091000 +0ms

eminberkayd, 2024-10-09:
Same issue is happing with Ubuntu 22.04. Nearly 6 months ago, I was able to get the discovered peripherals in my computer.

MrShinyAndNew, 2024-10-09:
I dug into this a bit further and I think there might be something else going on. Forgive me if I get this all wrong, as I haven't done C programming in years. But I wrote a sample program to try to replicate what Noble + HCI were doing and found that setsockopt would only work if the option passed in was sized as 16 bytes, not 14 as the setSocketFilter was doing. 

```struct hci_filter {
	uint32_t type_mask;
	uint32_t event_mask[2];
	uint16_t opcode;
};
... 
struct hci_filter hf;
printf("%d", sizeof (hf));
```
This outputs 16.
I changed hci.js's setFilter method to use a 16 byte buffer instead of 14 and changed the opcode to uint32 since I assume the problem is structure padding to align on word boundaries, and this got rid of the setsockopt error. 

Yay! Except that the program still doesn't work. The noble test program still does nothing and my node poweredup program still doesn't work. I noticed that many places in hci.js use non-multiples of 32 for the buffers to mimic the C structures, so I wonder if maybe that is the issue? But I'm just speculating as I'm in over my head here given the last time I programmed in C was in the 90s.

arenddeboer, 2024-11-04:
I had the same issue and decided to run the entire app in qemu/KVM virtual machine with Node.js 20 and Ubuntu 22. I redirect the USB Bluetooth dongle to the VM on boot. Maybe not the best easiest workaround but it does the job for now.