abandonware/noble #343: Doesn't work in the browser

maelp, 2024-10-17:
When trying to import a lib which itself uses noble from a Vue app, I get this error

[image]

maelp, 2024-10-17:
it seems one of the dependencies of `noble` is looking for the `process` variable which doesn't exist when using Vite (we need `import.meta.env`)

BFjacky, 2024-11-14:
This package is not for browser environment . Should only be used in Node.js. "Process" is one of core module of  NODE.JS

maelp, 2024-11-14:
Hmmmm it's too bad that it implements a wrapper for Webbluetooth but wouldn't be available in browsers no?

maelp, 2024-11-14:
If using `vite` you could just use `meta` rather than `process` to access ENV variable both in Node and browser

monteslu, 2025-02-05:
> This package is not for browser environment . Should only be used in Node.js. "Process" is one of core module of NODE.JS

It can be. Was browserifyable a few years ago at least:

https://github.com/noble/noble/pull/238