abandonware/noble #345: unknown state never passing to poweredOn

X4V1, 2024-11-14:
Hello,

I recently updated my debian 12 and after the update I'm not able to use my usb bluetooth adapter anymore.
Nothing has changed from code or anything else than the update of some packages.

After some tests I found that the noble.state is returning "unknown" and never changes to "poweredOn".
I checked the bluetooth adapter and it seems ok:
```
root@ws-api:~# hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: 00:15:83:EC:CC:A1  ACL MTU: 310:10  SCO MTU: 64:8
        UP RUNNING
        RX bytes:1416 acl:0 sco:0 events:97 errors:0
        TX bytes:4913 acl:0 sco:0 commands:97 errors:0
        Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: PERIPHERAL ACCEPT
        Name: 'ws-api'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 4.0 (0x6)  Revision: 0x22bb
        LMP Version: 4.0 (0x6)  Subversion: 0x22bb
        Manufacturer: Cambridge Silicon Radio (10)
```

The status of bluetooth in systemctl is also ok (not giving any warning/error):
```
root@ws-api:~# systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
    Drop-In: /etc/systemd/system/bluetooth.service.d
             └─experimental.conf
     Active: active (running) since Thu 2024-11-14 16:43:02 CET; 22s ago
       Docs: man:bluetoothd(8)
   Main PID: 1872 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 2308)
     Memory: 892.0K
        CPU: 21ms
     CGroup: /system.slice/bluetooth.service
             └─1872 /usr/libexec/bluetooth/bluetoothd --experimental

Nov 14 16:43:02 ws-api systemd[1]: Starting bluetooth.service - Bluetooth service...
Nov 14 16:43:02 ws-api bluetoothd[1872]: Bluetooth daemon 5.66
Nov 14 16:43:02 ws-api systemd[1]: Started bluetooth.service - Bluetooth service.
Nov 14 16:43:02 ws-api bluetoothd[1872]: Starting SDP server
Nov 14 16:43:02 ws-api bluetoothd[1872]: Bluetooth management interface 1.22 initialized
Nov 14 16:43:02 ws-api bluetoothd[1872]: Battery Provider Manager created
Nov 14 16:43:02 ws-api bluetoothd[1872]: Adv Monitor Manager created with supported features:0x00000000, enabled features:0x00000000, max number of supported monitors:32, max number of supported patterns:16
```

I don't see anything weird in the journal neither:
```
root@ws-api:~# journalctl -xe | grep bluetooth
Nov 14 16:44:42 ws-api systemd[1]: Starting bluetooth.service - Bluetooth service...
░░ Subject: A start job for unit bluetooth.service has begun execution
░░ A start job for unit bluetooth.service has begun execution.
Nov 14 16:44:42 ws-api bluetoothd[389]: Bluetooth daemon 5.66
Nov 14 16:44:42 ws-api bluetoothd[389]: Starting SDP server
Nov 14 16:44:42 ws-api systemd[1]: Started bluetooth.service - Bluetooth service.
░░ Subject: A start job for unit bluetooth.service has finished successfully
░░ A start job for unit bluetooth.service has finished successfully.
Nov 14 16:44:42 ws-api systemd[1]: Reached target bluetooth.target - Bluetooth Support.
░░ Subject: A start job for unit bluetooth.target has finished successfully
░░ A start job for unit bluetooth.target has finished successfully.
Nov 14 16:44:42 ws-api bluetoothd[389]: Bluetooth management interface 1.22 initialized
Nov 14 16:44:42 ws-api bluetoothd[389]: Battery Provider Manager created
Nov 14 16:44:42 ws-api dbus-daemon[390]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.4' (uid=0 pid=389 comm="/usr/libexec/bluetooth/bluetoothd --experimental")
Nov 14 16:44:42 ws-api bluetoothd[389]: Adv Monitor Manager created with supported features:0x00000000, enabled features:0x00000000, max number of supported monitors:32, max number of supported patterns:16
```

And from bluetoothctl I can connect to device using command line so I don't think the problem is coming from the adapter or the operating system configuration.

I'm running out of idea and I'm don't know what I can do next to investigate further. Any idea ?

I also found this opened issue #341 but I'm not sure if it is related to my problem.

Any help would be reallly appreciated.

mman, 2024-12-13:
Seeing this as well on my Raspberry Pi, downgrading to `6.6.51+rpt-rpi-v8` makes it work again.

X4V1, 2024-12-23:
It's now fixed (explained in the update 4). Closing this issue.

closed 2024-12-23 by X4V1

kbrawanski, 2024-12-29:
I can confirm that reverting to kernel version `6.6.51+rpt-rpi-v8` solves the problem. Does anyone know how to solve the problem on newer kernels?

btw, if anyone has a problem doing a revert, search for a commit from the repository `https://github.com/raspberrypi/rpi-firmware/commits/master/`, in this case to revert to `6.6.51+rpt-rpi-v8` do `sudo rpi-update d5a7dbe77b71974b9abb133a4b5210a8070c9284`.