abandonware/noble #356: Deprecate this project in favor of @stoprocent/noble because actively mainteined

Apollon77, 2025-04-22:
Hey All, Hey @rzr,

This project was great while noble was basically unmaintained and @rzr helped a lot providing this place to have a way to get important updates.

In the meantime there is also [@stoprocent/noble](https://github.com/stoprocent/noble) which is very actively maintained and also offers a way more improved version of noble already.

I want to recommend, please do not take that offensive, to deprecate this repo and library and move users to the new maintained noble branch, because this place is - in my eyes- no longer needed now and more confusing than helping.

@rzr Thanks a lot for all your work!!

Ingo

rzr, 2025-04-22:
no problem, ideally I would expect upstream to give the maintenance to an active maintainer, let me pin that issue anyway

stoprocent, 2025-04-22:
@rzr because it was taking too long to contribute

stoprocent, 2025-04-22:
I dont have an ambition to take over but I need this library for work and I need to release fast and often. 
Right now https://www.npmjs.com/package/@abandonware/noble is getting 2k downloads and https://www.npmjs.com/package/@stoprocent/noble is getting 12-15k and I think it makes people confused that's all

rzr, 2025-04-22:
ok fair enough, i wish it would be more sustainable if community maintained, ideally with upstream cooperation but let's face reality, according to your figures the "alone faster" flow  seems to suit better the users than  "together slower" ... history is not over though.

stoprocent, 2025-04-22:
Hey @rzr, it feels like there haven't been any significant improvements pushed to the library in quite a while — few bug fixes, no updates for newer ECMAScript features, and the release process has been quite slow. There are also many open issues that haven't received any response. I’m sorry, but I can't agree with the idea of continuing at this slower pace. I'm happy to move my fork under some common org and maintain it there, but you have to contribute something before saying history is not over. There is still tons to do under this lib.

rzr, 2025-04-22:
If you committed to takeover maintenance, I would be more than happy to move this repo under @adoptware org, there you can add your changes on top of it, but one question remains how can users trust you ? It is not personal but i was my main concerns to avoid hijacking of orphan projects... I thought about offloading trust to existing oss orgs but i have to say that does not scale well.

rzr, 2025-04-22:
Don't get me wrong this repo was created as a interim until someone is committed to maintain it in place of upstream. 

It looks like you did awesome job, and TBH you don't need me to keep doing what you have been doing so far.

If project lives in yours hands, great ! I also appreciate that you even took over some of @abandonware changes, to push the project forward.

The question I have today, how to make it scalable from a community perspective, to avoid the situation we faced with upstream. And how to keep the same level of trust ? If possible? Any thought ?

Meanwhile the issue is pinned in this org, so I hope you will get more inputs from here,
and if there is new PRs I will ping you or invite patches to be forward to the active maintainers (btw there should be several (too much) noble repo? how to deal with this fragmentation ?)

rzr, 2025-04-22:
> [@Apollon77](https://github.com/Apollon77) had an idea to move it to `noble-ng` org and keep it there. I'm ok with this, then I will maintain it until I can and in the meantime, we figure out the rules.

Let me ask do you already belonging to an existing established (trustworthy) OSS organization ?

If not, your proposal is fine, there is nothing i can do beside promoting it.

Understand my point, when I created this org I committed to provide a minimum level of trust, so I am reluctant to relax that now.

Apollon77, 2025-04-23:
@rzr I think we had that discussion already several times, I proofed my Open Source Software investments a lot I think and also @stoprocent is trustworthy.

I agree that trust is important, but these days trust is build by actions because they proof who you - are on GitHub at least and that's "the" place. Also the original noble maintainers were "just" some developers that build it and then decided for reasons (likely funding and flexibility in contributions and such) to create the "Noble" orga. Now they because unresponsive and all tries to contact them failed. So for whatever reason these 3-5 people became unresponsive in such a way and do not seem to be interested in seeing their project continuing to live and grow ... irrelevant because it simply is a fact.

And yes I tried to contact them several times within 1 year now and honestly ... With this I give up on that and find more practical ways to bring the project forward.

So I tried to join your project, but in fact as you know we did not came further. Someone decisions need to be made and #289 should have been an easy one and is not moving since 2 years ... including long times of discussions here and also the two of us aside. So also this approach does not work in practice to bring the project forward. All this blocks contributions and honestly Open Source s all about contributions.

So with this I favor the idea to start new.

I created GitHub orga "noble-ng" and we can move the stoprocent fork (because evolved from yours and enhanced a lot) and also packages into that. Then we use common processes there that are proofed in Open Source software development and how I also collaborate in a lot of other projects that are working.

I invite you to also join there - or any of the OSS orgs you mentioned above - to bring Noble and Bleno and connected projects forward and offer a stable and reliable Bluetooth stack for the Node.js development community - because that's it is all about.